sunfounder/pironman5 #1: Support request for DietPi operating system.

Krouwndouwn, 2024-05-23:
DietPi is an extremely lightweight Debian OS, highly optimised for minimal CPU and RAM resource usage, ensuring your SBC always runs at its maximum potential.

Website: https://dietpi.com/
GitHub: https://github.com/MichaIng/DietPi

Tartouffe, 2025-02-19:
Its working under DietPi, just installed it in that way, minutes ago:
> sudo apt-get update
> sudo apt-get install git -y
> sudo apt-get install python3 python3-pip python3-setuptools -y

# Install Pironman Module
> cd ~
> git clone https://github.com/sunfounder/pironman5
> cd ~/pironman
> sudo python3 install.py

REBOOT

Now the backwards FANs (GPIO Fans) are running loud and the OLED shows its stuff. 

You can set speed, color etc. with:
> http://[IP of the Pironman unit]:34001 

At the top right corner you will see a windows-like icon. Click onto that. 
Set the **GPIO Fan Mode** to any other value than 'always on'. 
And any other 'Pironman' values.

Krouwndouwn, 2025-02-25:
@Tartouffe 

Im running: DietPi (RPi 5 Model B (aarch64)) v9.11.2 (Bookworm)

**I did try the commands below:** (from: https://docs.sunfounder.com/projects/pironman5/en/latest/set_up/set_up_rpi_os.html)

```
sudo apt-get update
sudo apt-get install git -y
sudo apt-get install python3 python3-pip python3-setuptools -y
```
(Differs a little from your commands u mentioned...)

```
cd ~
git clone https://github.com/sunfounder/pironman5.git
cd ~/pironman5
sudo python3 install.py
```
(Differs a little from your commands u mentioned...)

But after the install command failed:

```
root@DietPi:~/pironman5# sudo python3 install.py
Installing for Pironman 5

Install build dependencies...
 - Update package list... Done
 - Install build dependencies: python3-pip python3-venv git curl... Done
Run commands before install...
 - Run command before install: Install LGPIO... Done
 - Run command before install: Download influxdb key... Done
 - Run command before install: Setup influxdb install source... Done
 - Run command before install: Cleanup influxdata-achive.key... Done
Install APT dependencies...
 - Install APT dependencies: i2c-tools libopenjp2-7 kmod influxdb libfreetype6-dev lsof libjpeg-dev ... Done
Create working directory...
 - Create work directory... Done
 - Change work directory mode... Done
 - Change work directory owner... Done
 - Create log directory... Done
 - Change log directory mode... Done
 - Change log directory owner... Done
 - Create virtual environment... Done
Install Python source packages...
 - Uninstall pironman5 old package... Done
 - Install pironman5 from source... Error
 - Uninstall pm_auto old package... Done
 - Install pm_auto from source... Error
 - Uninstall pm_dashboard old package... Done
 - Install pm_dashboard from source... Done
 - Uninstall sf_rpi_status old package... Done
 - Install sf_rpi_status from source... Done
Setup auto start...
 - Copy binary file... Done
 - Change binary file mode... Done
 - Copy service file... Done
 - Enable service... Done
 - Reload systemd... Done
Probe modules...
 - add module: i2c-dev... Done
Copy device tree overlay...
 - Copy dtoverlay sunfounder-pironman5.dtbo... Done
Change work directory owner...
 - Change work directory owner... Done
Finished
Cleanup
 - Remove build... Done

[... 399 lines cut ...]
      running install
      /opt/pironman5/venv/lib/python3.11/site-packages/setuptools/command/install.py:34: Setuptools DeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based too ls.
        warnings.warn(
      running build
      running build_ext
      building 'sysv_ipc' extension
      creating build
      creating build/temp.linux-aarch64-cpython-311
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-stron g -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/opt/pironman5/venv/include -I/usr/includ e/python3.11 -c common.c -o build/temp.linux-aarch64-cpython-311/common.o
      error: command 'aarch64-linux-gnu-gcc' failed: No such file or directory
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> sysv-ipc

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

Try to fix it yourself, or contact service@sunfounder.com with this message
```

If u know how to get this fixed please let me know.

db1488, 2025-05-03:
Looks like it's trying to build C extensions for the python packages, and the toolchain doesn't exist. What worked for me was to run:

`sudo apt install build-essential python3-dev python3-pip gcc g++ git`

before running `install.py` from the Pironman 5 install docs. Hope this helps!

db1488, 2025-05-03:
> [@Tartouffe](https://github.com/Tartouffe)
> 
> Im running: DietPi (RPi 5 Model B (aarch64)) v9.11.2 (Bookworm)
> 
> **I did try the commands below:** (from: https://docs.sunfounder.com/projects/pironman5/en/latest/set_up/set_up_rpi_os.html)
> 
> ```
> sudo apt-get update
> sudo apt-get install git -y
> sudo apt-get install python3 python3-pip python3-setuptools -y
> ```
> 
> (Differs a little from the commands u mentioned...)
> 
> ```
> cd ~
> git clone https://github.com/sunfounder/pironman5.git
> cd ~/pironman5
> sudo python3 install.py
> ```
> 
> (Differs a little from the commands u mentioned...)
> 
> But after the install command it failed:
> 
> ```
> root@DietPi:~/pironman5# sudo python3 install.py
> Installing for Pironman 5
> 
> Install build dependencies...
>  - Update package list... Done
>  - Install build dependencies: python3-pip python3-venv git curl... Done
> Run commands before install...
>  - Run command before install: Install LGPIO... Done
>  - Run command before install: Download influxdb key... Done
>  - Run command before install: Setup influxdb install source... Done
>  - Run command before install: Cleanup influxdata-achive.key... Done
> Install APT dependencies...
>  - Install APT dependencies: i2c-tools libopenjp2-7 kmod influxdb libfreetype6-dev lsof libjpeg-dev ... Done
> Create working directory...
>  - Create work directory... Done
>  - Change work directory mode... Done
>  - Change work directory owner... Done
>  - Create log directory... Done
>  - Change log directory mode... Done
>  - Change log directory owner... Done
>  - Create virtual environment... Done
> Install Python source packages...
>  - Uninstall pironman5 old package... Done
>  - Install pironman5 from source... Error
>  - Uninstall pm_auto old package... Done
>  - Install pm_auto from source... Error
>  - Uninstall pm_dashboard old package... Done
>  - Install pm_dashboard from source... Done
>  - Uninstall sf_rpi_status old package... Done
>  - Install sf_rpi_status from source... Done
> Setup auto start...
>  - Copy binary file... Done
>  - Change binary file mode... Done
>  - Copy service file... Done
>  - Enable service... Done
>  - Reload systemd... Done
> Probe modules...
>  - add module: i2c-dev... Done
> Copy device tree overlay...
>  - Copy dtoverlay sunfounder-pironman5.dtbo... Done
> Change work directory owner...
>  - Change work directory owner... Done
> Finished
> Cleanup
[... 408 lines cut ...]
>       creating build
>       creating build/temp.linux-aarch64-cpython-311
>       aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-stron g -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/opt/pironman5/venv/include -I/usr/includ e/python3.11 -c common.c -o build/temp.linux-aarch64-cpython-311/common.o
>       error: command 'aarch64-linux-gnu-gcc' failed: No such file or directory
>       [end of output]
> 
>   note: This error originates from a subprocess, and is likely not a problem with pip.
> error: legacy-install-failure
> 
> × Encountered error while trying to install package.
> ╰─> sysv-ipc
> 
> note: This is an issue with the package mentioned above, not pip.
> hint: See above for output from the failure.
> 
> Try to fix it yourself, or contact service@sunfounder.com with this message
> ```
> 
> If u know how to get this fixed please let me know.

Looks like it's trying to build C extensions for the python packages, and the toolchain doesn't exist. What worked for me was to run:

`sudo apt install build-essential python3-dev python3-pip gcc g++ git`

before running `install.py` from the Pironman 5 install docs. Hope this helps!

Krouwndouwn, 2025-06-02:
@db1488 

Thank you for your response! I tested your solution, and it finally seems to be correctly installed and working on `DietPi (RPi 5 Model B (aarch64)) v9.13.2 (Bookworm)`!!

I followed the steps from the official docs: https://docs.sunfounder.com/projects/pironman5/en/latest/pironman5/set_up/set_up_rpi_os.html#downloading-and-installing-the-pironman5-module . I used your code for step 1, while for step 2, I followed the official procedure with an adjusted/latest pironman5 software version.

## Steps taken

**Step 1:**
```
dietpi-update
sudo apt install build-essential python3-dev python3-pip gcc g++ git -y
```

**Step 2:**
```
cd ~
git clone https://github.com/sunfounder/pironman5.git -b 1.2.9
cd ~/pironman5
sudo python3 install.py
```

Hopefully, this approach will help other DietPi users as well. It would be great to see official support added in the future.

However, I’ve noticed a few issues:
- A 404 error appears when opening the dashboard.
- Several warnings and errors are visible in the logs.
- Some history items can’t be accessed due to "unsupported mean iterator type: query.stringInterruptIterator" errors.

It seems that the software isn’t fully compatible with DietPi yet, or there are some bugs in the latest version. Hopefully, these will be resolved in future updates.

Thanks again!