sunfounder/pironman5 #7: Pironman takes up a whole core

Willy65, 2024-10-06:
When pironman is started, it consumes 100% CPU time on 1 core.
See output below from top command, after issuing "pironman start" from the commandline (as root):

```
> ```
> top - 21:33:05 up 43 min,  4 users,  load average: 1.77, 0.87, 0.44
> Tasks: 194 total,   2 running, 192 sleeping,   0 stopped,   0 zombie
> %Cpu0  :  0.0 us,  0.3 sy,  0.0 ni, 98.3 id,  1.0 wa,  0.0 hi,  0.3 si,  0.0 st 
> %Cpu1  : 51.3 us, 48.7 sy,  0.0 ni,  0.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st 
> %Cpu2  :  0.0 us,  0.3 sy,  0.0 ni, 99.7 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st 
> %Cpu3  :  0.0 us,  0.0 sy,  0.0 ni,100.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st 
> MiB Mem :   7945.1 total,   7029.1 free,    356.1 used,    675.2 buff/cache     
> MiB Swap:    200.0 total,    200.0 free,      0.0 used.   7589.0 avail Mem 
> 
>   PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                       
>  1823 root      20   0   68720  30804  14336 R  99.7   0.4   2:25.95 python3                                                                                                       
>     1 root      20   0   35900   9776   7916 S   0.0   0.1   0:00.86 systemd                                                                                                       
>     2 root      20   0       0      0      0 S   0.0   0.0   0:00.00 kthreadd                                                                                                      
> ```
```

If this is just for displaying CPU temperature or fan speed on the OLED display, I'm not happy.

Lovmong, 2024-11-06:
We have not reproduced your problem, and the usage we tested is around 5%. Are you using the latest program? Are there any other programs running that cause conflicts?
[image]
[image]

nairn62, 2025-02-27:
cd ~
git clone https://github.com/sunfounder/pironman.git -b v2.0
cd ~/pironman
sudo python3 install.py

This code works, but one CPU core is constantly at 100%.

MrBasa, 2025-05-12:
I have seen the same behavior, but only when viewing the Pironman stat page. Maybe it's due to the query behavior to 'influxd'? I'll try to follow up with my local stats.

Willy65, 2025-05-13:
Sorry for the late response, I screwed up with my MFA.
I have a 2nd Pironman now, showing similar behaviour.
I'll try to reproduce my setup, will keep you posted.

Sjain-dir, 2025-05-16:
Hey, I am also facing the same issue, is there any update ?

MrBasa, 2025-05-16:
Things look chill when the service is just running:

[image]

Here's how things look when I'm viewing the stat page:

[image]

Sorry, I had Folding At Home running there, should have shut that down to eliminate noise, but I think it's quite visible the behavior of the Pironman service when viewing the stat page vs not.

MrBasa, 2025-05-16:
I don't have any other services running that use influxd, so the only thing that could be hammering it is this Pironman service.

MrBasa, 2025-05-16:
If you're experiencing the same symptoms, I'd recommend just leaving the service running if you want to have the case display functioning, but not visit the stat web page.

MrBasa, 2025-05-17:
> We have not reproduced your problem, and the usage we tested is around 5%. Are you using the latest program? Are there any other programs running that cause conflicts? [image] [image]

@Lovmong  Were you visiting the stat page while checking these diagnostics? I think the issue, at least as I've experienced it, only occurs when visiting the stat page.

MrBasa, 2025-05-17:
Sorry, some additional context that I didn't give previously is that it's not the service itself that I see spiking, it's Influx. My basic web search informed me that it's a database, but I don't know anything about it, I just see its usage spike anytime I visit the page hosted by the Pironman service. Hopefully that will help identify the issue. Seems to me that the Pironman service is hammering the database or is using a very suboptimal query.

cavonlee, 2025-08-04:
It seems like because of the read and set function is interferes of the influxdb. we have fix it in newest 1.3.x branch. 1.3 version will release soon.