sunfounder/pironman5 #12: prironman5 funktioniert nicht mehr nach Update HA OS 14

Matthias1977-tech, 2024-12-04:
Nach dem Update auf Homeassistant OS 14 funktioniert prionman5 (1.1.1) nicht mehr.

Rollback auf Version 13.2 funktioniert wieder!

rsocko, 2024-12-05:
I am seeing the same issue. I thought at first it was the Pironman 1.1.1 update. So I also tried reverting to the earlier Pironman addon version, but that didn't help. Going to try to revert the HASS OS like OP.

ad-cdavila, 2024-12-05:
@rsocko I tried the same as you. What it worked for me was downgrading home assistant OS to 13.2. Seems like it's related to a kernel update in the new HA version

rsocko, 2024-12-05:
thank you @ad-cdavila The rollback to OS 13.2 also worked to solve it for me. Learning all kinds of new benefits of keeping backups, and roll back commands for the OS.

Matthias1977-tech, 2024-12-06:
Hallo, die Version 1.1.2 funktioniert jetzt mit HA 14.0

closed 2024-12-06 by Matthias1977-tech

MHx489, 2024-12-15:
> Hallo, die Version 1.1.2 funktioniert jetzt mit HA 14.0

Wie hast du es hinbekommen? Ich habe immer noch das Problem, dass das Addon gar nicht startet. Siehe Log was ich im Issue #17 angehängt habe. Ich habe die neuste HA Version und Addon 1.1.2