sunfounder/pironman5 #13: Settings do not persist between reboots. Where is config.json?

jkreienh, 2024-12-12:
### My Issue:

- I'm running HA 14.0 on a pi5
- I'm able to change the FAN, LED and Temperature Units via the WEB UI. 
- The settings do not persist between reboots.  

**Note: Within the WEB UI, the save button is no longer available.**

[image]

**Where is the config.json file stored that I can manually update?** 

The logs do not show the location of this file when running the HA OS install.

[image]

Source code assumes a pironman5/config.json file, but I'm not sure where to find it on my system:
[image]

jkreienh, 2025-01-09:
@cavonlee -  I'm not sure where this file would exist within my HA PI5 filesystem.  Can you provide the actual path?

LYN3S-CO-UK, 2025-01-14:
I wanted to know the same to disabled the OLED, as does not appear to be a command option. 

It was located alongside the pironman5.py in my Ubuntu install

` sudo find / -name "*pironman5.py"`

`/opt/pironman5/venv/lib/python3.12/site-packages/pironman5/config.json
`

jkreienh, 2025-01-15:
@LYN3S-CO-UK 

I'm running HA on the PI5 Supervised so I don't think this applies.

jkreienh, 2025-01-15:
----------------------------------------------------------
 Add-on: Pironman 5
 Pironman 5 support
-----------------------------------------------------------
 Add-on version: 1.1.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.1  (aarch64 / raspberrypi5-64)
 Home Assistant Core: 2025.1.0
 Home Assistant Supervisor: 2024.12.3
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[09:22:34] INFO: Configuring timezone (America/Chicago)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
25/01/15 09:22:35.181 [WARNING] Failed to create gpiochip4 symlink
25/01/15 09:22:35.196 [DEBUG] Initializing OLED
25/01/15 09:22:35.267 [DEBUG] Update temperature_unit to C
25/01/15 09:22:35.268 [DEBUG] Update oled_enable to True
25/01/15 09:22:35.268 [DEBUG] Update oled_rotation to 0
25/01/15 09:22:35.268 [DEBUG] Update oled_disk to total
25/01/15 09:22:35.268 [DEBUG] Update oled_network_interface to all
25/01/15 09:22:35.268 [DEBUG] OLED initialized
25/01/15 09:22:35.268 [DEBUG] Update LED count: 4
25/01/15 09:22:35.268 [DEBUG] Update LED count: 4
25/01/15 09:22:35.268 [DEBUG] Update RGB enable: True
25/01/15 09:22:35.269 [DEBUG] Update RGB enable: True
25/01/15 09:22:35.269 [DEBUG] Update RGB color: #0a1aff
25/01/15 09:22:35.269 [DEBUG] Update RGB color: [10, 26, 255]
25/01/15 09:22:35.269 [DEBUG] Update RGB brightness: 50
25/01/15 09:22:35.269 [DEBUG] Update RGB brightness: 50
25/01/15 09:22:35.269 [DEBUG] Update RGB speed: 50 AKA 50
25/01/15 09:22:35.269 [DEBUG] Update RGB style: breathing
25/01/15 09:22:35.269 [DEBUG] Update RGB style: breathing
25/01/15 09:22:35.282 [DEBUG] WS2812 initialized
25/01/15 09:22:35.283 [DEBUG] FanControl init
25/01/15 09:22:35.283 [DEBUG] FAN update_config: {'rgb_color': '#0a1aff', 'rgb_brightness': 50, 'rgb_style': 'breathing', 'rgb_speed': 50, 'rgb_enable': True, 'rgb_led_count': 4, 'temperature_unit': 'C', 'oled_enable': True, 'oled_rotation': 0, 'oled_disk': 'total', 'oled_network_interface': 'all', 'gpio_fan_pin': 6, 'gpio_fan_mode': 2, 'gpio_fan_led': 'follow', 'gpio_fan_led_pin': 5}
25/01/15 09:22:35.283 [DEBUG] Update gpio_fan_pin to 6
25/01/15 09:22:35.283 [DEBUG] Update gpio_fan_mode to 2
25/01/15 09:22:35.283 [DEBUG] Update gpio_fan_led to follow
25/01/15 09:22:35.284 [DEBUG] Update gpio_fan_led_pin to 5
25/01/15 09:22:35.284 [DEBUG] Init GPIO Fan with pin: 6
/usr/lib/python3/dist-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: 'can not open gpiochip'
  warnings.warn(
25/01/15 09:22:35.315 [DEBUG] Init PWM Fan
25/01/15 09:22:35.320 [INFO] SPC peripheral disabled
25/01/15 09:22:35.320 [INFO] Setting debug level to INFO
25/01/15 09:22:35.321 [INFO] PM Auto Start
25/01/15 09:22:35.321 [INFO] PMAuto started
25/01/15 09:22:35.327 [INFO] Starting influxdb service
ts=2025-01-15T15:22:35.333994Z lvl=info msg="InfluxDB starting" log_id=0u6dnFmG000 version=1.6.7~rc0 branch=unknown commit=unknown
ts=2025-01-15T15:22:35.334021Z lvl=info msg="Go runtime" log_id=0u6dnFmG000 version=go1.22.2 maxprocs=4
ts=2025-01-15T15:22:35.351814Z lvl=info msg="Using data dir" log_id=0u6dnFmG000 service=store path=/var/lib/influxdb/data
ts=2025-01-15T15:22:35.351970Z lvl=info msg="Compaction settings" log_id=0u6dnFmG000 service=store max_concurrent_compactions=2 throughput_bytes_per_second=50331648 throughput_burst_bytes=50331648
ts=2025-01-15T15:22:35.351990Z lvl=info msg="Open store (start)" log_id=0u6dnFmG000 service=store trace_id=0u6dnFql000 op_name=tsdb_open op_event=start
ts=2025-01-15T15:22:35.352055Z lvl=info msg="Open store (end)" log_id=0u6dnFmG000 service=store trace_id=0u6dnFql000 op_name=tsdb_open op_event=end op_elapsed=0.066ms
ts=2025-01-15T15:22:35.352092Z lvl=info msg="Opened service" log_id=0u6dnFmG000 service=subscriber
ts=2025-01-15T15:22:35.352100Z lvl=info msg="Starting monitor service" log_id=0u6dnFmG000 service=monitor
ts=2025-01-15T15:22:35.352105Z lvl=info msg="Registered diagnostics client" log_id=0u6dnFmG000 service=monitor name=build
ts=2025-01-15T15:22:35.352109Z lvl=info msg="Registered diagnostics client" log_id=0u6dnFmG000 service=monitor name=runtime
ts=2025-01-15T15:22:35.352113Z lvl=info msg="Registered diagnostics client" log_id=0u6dnFmG000 service=monitor name=network
ts=2025-01-15T15:22:35.352122Z lvl=info msg="Registered diagnostics client" log_id=0u6dnFmG000 service=monitor name=system
ts=2025-01-15T15:22:35.352143Z lvl=info msg="Starting precreation service" log_id=0u6dnFmG000 service=shard-precreation check_interval=10m advance_period=30m
[... 17 lines cut ...]
25/01/15 09:22:37.379 [INFO] Dashboard Server start
25/01/15 09:22:37.384 [DEBUG] Checking if influxdb is ready...
[httpd] ::1 - root [15/Jan/2025:09:22:37 -0600] "GET /ping HTTP/1.1" 204 0 "-" "python-requests/2.32.3" 8d59c7ef-d354-11ef-8004-000000000000 39
25/01/15 09:22:37.396 [INFO] Influxdb is ready
[... cut ...]
[httpd] ::1 - root [15/Jan/2025:09:22:45 -0600] "GET /ping HTTP/1.1" 204 0 "-" "python-requests/2.32.3" 91e8b9f6-d354-11ef-8014-000000000000 25
[httpd] ::1 - root [15/Jan/2025:09:22:45 -0600] "POST /write?db=pironman5 HTTP/1.1" 204 0 "-" "python-requests/2.32.3" 91e8fe88-d354-11ef-8015-000000000000 15014
[httpd] ::1 - root [15/Jan/2025:09:22:46 -0600] "GET /ping HTTP/1.1" 204 0 "-" "python-requests/2.32.3" 928ce910-d354-11ef-8016-000000000000 24
[httpd] ::1 - root [15/Jan/2025:09:22:46 -0600] "POST /write?db=pironman5 HTTP/1.1" 204 0 "-" "python-requests/2.32.3" 928d30f5-d354-11ef-8017-000000000000 1

physicsG, 2025-01-30:
For me the settings are not persisted either; every time the Pi5 reboots, the LEDs stay on and I have to go into the settings to disable them.

cavonlee, 2025-08-04:
the newest version fix the problem, try a update

closed 2025-08-04 by cavonlee