sunfounder/pironman5 #25: Crash on boot when FTDI USB cable is connected: tries to access missing gpiochip14

marcelkooijman, 2025-04-17:
**Bug Description**  
When booting a Raspberry Pi 5 with a Pironman 5 case, the `pironman5` systemd service fails to start **if a USB-to-serial FTDI cable is connected** during boot. The following error appears in the logs:

```
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/gpio/devices/gpiochip14/of_node/compatible'
```

This issue does **not** occur if the FTDI cable is unplugged during boot — then the service starts without problems.

---

**What’s Likely Going Wrong**  
It seems that `pironman5` is attempting to access `gpiochip14` directly, which **does not exist** on the Raspberry Pi 5. The FTDI USB cable may be triggering some detection logic that incorrectly assumes the presence of this GPIO chip.

Example of available GPIO chips on this system:
```
gpiochip512
gpiochip529
gpiochip535
gpiochip567
gpiochip571
gpiochip625
```

---

**Steps to Reproduce**
1. Plug in a USB-to-serial FTDI cable (idVendor=0403, idProduct=6001).
2. Reboot the Raspberry Pi 5.
3. Observe that `pironman5.service` fails to start.
4. Unplug the cable and reboot — now the service starts fine.

---

**Expected Behavior**  
The service should gracefully handle missing GPIO chips and avoid hardcoded references to non-existent ones like `gpiochip14`.

---

**System Info**
- Raspberry Pi 5
- Pironman 5 case
- OS: Raspberry Pi OS (64-bit)
- Service: `pironman5` installed via systemd
- FTDI cable (0403:6001) connected during boot

---

Klaar om te plakken in GitHub dus! Laat me gerust weten als je je gebruikersnaam erbij wilt zetten of nog iets wil toevoegen 👌

bootyberry, 2025-07-24:
This was a bug in adafruit blinka. A fix was merged upstream. 
I fixed it for now by manually patching this file with the commit.

/opt/pironman5/venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller/generic_linux/lgpio_pin.py

https://github.com/adafruit/Adafruit_Blinka/pull/967/commits/c50b9b5fd8fb6b28fc0ba9d928aa1a80d17724d9

marcelkooijman, 2025-07-26:
Thank you! The problem is gone.