sunfounder/pironman5 #26: Security - Service(s), Permissions, and Logging

maxwell2011, 2025-05-05:
Hey guys, I know software is hard, and I love the pironman5. 
Two things, because I imagine a lot of units wind up in the hands of people that would never be expected to 'know better.' 
`1` The package runs as root while making extensive use of `requests`. This would be less of a problem if it wasn't touching bare metal, with access to system peripherals, like an integrated camera, (usually) bluetooth/wifi, and anything else the user adds. Assuming a standard user is working from minimum viable product router like the one coming from their ISP, or any consumer-grade all-in-one from the local big-box store, this can become a source of system failure. A service user + group, like `pironman5` with no home directory or password, and only read/execute permissions for `/usr/local/bin/pironman5` is _probably_ the right answer for 90% of the population.

`2` The [systemd](https://github.com/sunfounder/pironman5/blob/main/bin/pironman5.service) service defaults to `debug` logging (See `Line 10`, `ExecStart=...`). This is unnecessary in almost every case and directly related to `1`. Setting logging to `error` or `warning` would be the "right" answer here. For context, debug logging generates north of **70MB** of system logs _every day_. Continuous churn of logs like this can kill an SD card, wear down an SSD/NVMe drive, or just overall slow down the process of ingesting/checking/debugging other services. Even `sed` slowed down while stripping out the noise. 

```bash
# Logs stripped with the following:
journalctl -S yesterday | sed 's/^.*pironman5.*$/ /' | sed -r '/^\s*$/d' > clean-logs.txt
```
The relationship between `1` and `2` is that a lot of users run essentially unmodified versions of their OS of choice, and logging at massive volumes without cleanup mechanisms can (and does) take down systems.

Additional Suggestions:
Simple user auth or token mechanism, and an automatic timeout disable feature to turn it off after `x` days, eg: auto disable after a week or two without activity, and a notice showing a link to a help doc with instructions on how to reactivate from the system itself.
A simple drop down to flip the log level to the desired level, or an automatic edit to the service config file after a week or three to turn down logging from `debug` to `error`.

maxwell2011, 2025-05-14:
Serious additional problems include `SF_Installer.create_working_dir`, and the abuse of `chmod +777` when creating folders.

MrBasa, 2025-05-14:
Lol... These guys...

MrBasa, 2025-05-14:
I don't have the for just giving us home.

cavonlee, 2025-05-15:
@maxwell2011 Great, thanks for your suggestion, we will look into it.