sunfounder/pironman5 #28: Pironman5 cannot find adafruit_blinka

jscheidtmann, 2025-06-19:
I have installed pironman5 as described, but the services startup fails:

```
grimaldi@pironman5:~ $ uname -a 
Linux pironman5 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
grimaldi@pironman5:~ $ cat /etc/debian_version 
12.11
grimaldi@pironman5:~ $ sudo journalctl -b -u pironman5
Jun 19 13:56:36 pironman5 systemd[1]: Starting pironman5.service - pironman5 service...
Jun 19 13:56:36 pironman5 systemd[1]: Started pironman5.service - pironman5 service.
Jun 19 13:56:37 pironman5 pironman5[1518]: Traceback (most recent call last):
Jun 19 13:56:37 pironman5 pironman5[1518]:   File "/opt/pironman5/venv/bin/pironman5-service", line 8, in <module>
Jun 19 13:56:37 pironman5 pironman5[1518]:     sys.exit(main())
Jun 19 13:56:37 pironman5 pironman5[1518]:              ^^^^^^
Jun 19 13:56:37 pironman5 pironman5[1518]:   File "/opt/pironman5/venv/lib/python3.11/site-packages/pironman5/__init__.py>
Jun 19 13:56:37 pironman5 pironman5[1518]:     from pm_auto.ws2812 import RGB_STYLES
Jun 19 13:56:37 pironman5 pironman5[1518]:   File "/opt/pironman5/venv/lib/python3.11/site-packages/pm_auto/ws2812.py", l>
Jun 19 13:56:37 pironman5 pironman5[1518]:     import board
Jun 19 13:56:37 pironman5 pironman5[1518]:   File "/opt/pironman5/venv/lib/python3.11/site-packages/board.py", line 21, i>
Jun 19 13:56:37 pironman5 pironman5[1518]:     from adafruit_blinka.agnostic import board_id, detector
Jun 19 13:56:37 pironman5 pironman5[1518]: ModuleNotFoundError: No module named 'adafruit_blinka'
Jun 19 13:56:37 pironman5 systemd[1]: pironman5.service: Main process exited, code=exited, status=1/FAILURE
Jun 19 13:56:37 pironman5 systemd[1]: pironman5.service: Failed with result 'exit-code'.
```

I tried reinstalling adafruit_blinka by pip uninstalling/install Adafruit-Blinka and Adafruit-Blinka, to no avail. I even tried using the official way for installing Adafruit Blinka (involving downloading a script), but I was not able to get rid of the error. 

Any help is appreciated.

jscheidtmann, 2025-06-19:
Seems to an issue with the Adafruit-Blinka package: https://github.com/adafruit/Adafruit_Blinka/issues/975, there already seems to be a suggested patch: https://github.com/adafruit/Adafruit_Blinka/issues/976

jscheidtmann, 2025-06-19:
Downgrading Adafruit-Blinka in the venv the pironman service runs helps. Run:

`/opt/pironman5/venv/bin/pip install Adafruit-Blinka==8.59.0`

and start the service:

`sudo systemctl start pironman5`

clevins69, 2025-06-19:
@jscheidtmann That did it,  Thank you!

RogueHammerite, 2025-06-19:
@jscheidtmann I was coming here to make the same suggestion except I ran into another odd issue. I uninstalled it, and chose the older version like you did, and at first it worked, but the I/O HAT+ 4 led's would not function. So I uninstalled adafruit_blinka and ran 

`sudo /opt/pironman5/venv/bin/pip uninstall adafruit_blinka ; sudo /opt/pironman5/venv/bin/pip install --upgrade \ git+https://github.com/adafruit/Adafruit_Blinka.git`

I also was missing the /opt/pironman5/venv/lib/python3.11/site-packages/microcontroller directory, so I tried to simply symlink the dir from "/opt/pironman5/venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller" but it failed miserably. So I removed the symlink "/opt/pironman5/venv/lib/python3.11/site-packages/microcontroller" and hand created it with

`cd /opt/pironman5/venv/lib/python3.11/site-packages/; mkdir microcontroller; touch microcontroller/__init__.py; ln -s ../adafruit_blinka/microcontroller/bcm2712/pin.py microcontroller/pin.py`

At this point I rebooted, and everything came up clean...

jscheidtmann, 2025-06-19:
In the mean time a new version was released. And pip using git will probably pick up the new version anyway.

RogueHammerite, 2025-06-20:
I don't doubt that one bit. Murphy's Razor, It's inevitable that a person beats themselves up while falling down the stairs trying to solve a problem only for a solution to come out just before they figured it out.

closed 2025-06-20 by jscheidtmann

jamb1955, 2025-06-22:
Hi, similar to my comment on #975.
I did as suggested above, but in my case it did not "come up clean" 
Still error on import board.

Any help is appreciated.

jscheidtmann, 2025-06-22:
The error message is different: „Could not determine Jetson model“.
Tomorrow I should be able to provide a list of packages & versions that work for me(tm)