sunfounder/pironman5 #31: DB Maintenace and Records retention

asheimo, 2025-06-22:
this project uses influxdb to store records for viewing history of the device. No where in the installation are maintenance plans or retention limits put in place. This leaves the device vulnerable to having the storage device being completely filled. 

I have been running my Pironman5 system for 6mo and the db is 3.2GB. While this doesn't seem like an issue, I have had a couple of occasions of the tempdb running wild and filling up the disk.

please implement maintenance and retention limits to the installation.

cavonlee, 2025-07-16:
Great suggestions, we will work on that.

cavonlee, 2025-07-28:
It's added to 1.3.x branch. Please wait for the stable release for 1.3 version.