sunfounder/pironman5 #32: Dashboard Request Error: 404 on get-data

jrbcastro, 2025-07-08:
I'm getting a popup error Request Error: 404 and found the error is coming from http://192.168.12.195:34001/api/v1.0/get-data

jrbcastro, 2025-07-22:
Attached are the log files.

[main.log](https://github.com/user-attachments/files/21358225/main.log)
[pm_auto.fan_control.log](https://github.com/user-attachments/files/21358223/pm_auto.fan_control.log)
[pm_auto.oled.log](https://github.com/user-attachments/files/21358222/pm_auto.oled.log)
[pm_auto.pm_auto.log](https://github.com/user-attachments/files/21358220/pm_auto.pm_auto.log)
[pm_dashboard.data_logger.log](https://github.com/user-attachments/files/21358221/pm_dashboard.data_logger.log)
[pm_dashboard.database.log](https://github.com/user-attachments/files/21358224/pm_dashboard.database.log)
[pm_dashboard.pm_dashboard.log](https://github.com/user-attachments/files/21358219/pm_dashboard.pm_dashboard.log)
[pm_auto.ssd1306.log](https://github.com/user-attachments/files/21358227/pm_auto.ssd1306.log)
[pm_auto.ws2812.log](https://github.com/user-attachments/files/21358226/pm_auto.ws2812.log)

cavonlee, 2025-07-28:
1.2.17 version is not stable, please try 1.2.15 version:

```bash
cd ~/pironman5
git checkout -b 1.2.15
sudo python3 install.py
```

puneethr09, 2025-07-28:
this still installs 1.2.17

I would use :
git fetch --tags
git checkout refs/tags/1.2.15

jrbcastro, 2025-07-30:
@cavonlee @puneethr09 thank you both. I will try and install version 1.2.15